sermatec-opensource/homeassistant-sermatec-inverter #9: Login in..

WitoldPalowski, 2022-09-20:
Hi, 
I've used your way to integrate Sermatec. I can see Sermatec Inverter in possible integrations in HA, but when I choose it, I am stucked now here. What should be filled in?
[image]

andreondra, 2022-09-22:
Hello, thank you for your report. Please, try the latest version, it is probably an issue with language files. First input box is an IP address of your inverter and the second is a port, you can fill in the default, which is 8899.

If the issue persists, please let me know.

WitoldPalowski, 2022-09-25:
Thanks. I can see already host = IP and port. It is important to remark, it is inverter's IP of STxxxx wifi. In my case 10.10.100.254. And I had to set up my WLAN connection on HA properly first. HA is set up that Cable LAN is on my main network and WLAN on inverter's wifi STxxxx.

closed 2022-09-25 by WitoldPalowski

andreondra, 2022-09-30:
Actually it is not needed to connect to the inverter's WiFi (although it is possible of course). If you connected the inverter to your home WiFi (the embedded UART bridge can work both in station and AP modes), it will be available in the local network for connection. Then the inverter's IP is determined by the local DHCP server.