sermatec-opensource/homeassistant-sermatec-inverter #10: Add integration to HACS

andreondra, 2022-09-24:
Depends on #6.

mathieupotier, 2023-01-18:
Following this doc, but I cannot consider this custom_component to be in stable lifecycle... better to keep working a little bit on issues.

https://hacs.xyz/docs/publish/include

andreondra, 2023-01-18:
I agree, this custom component is in an early alpha stage, I'd like to publish the component when we have all the basic functionality reliably implemented. Also, the logo requirement (#6) could be a little tricky to fulfill, because there could be licensing issues.

mathieupotier, 2023-01-18:
Sorry I can't help about the logo ... It's not my best skill ... ^_^

But I assume that we can use the [Sermatec logo](https://eur.sermatec-cloud.com/static/favicon.ico) ... but I'm not sure about the legal part ...

mathieupotier, 2024-06-02:
I'll try to generate a logo with some AI (stable diffusion) got few nice results in another project...
Or maybe the logo of this GitHub organisation... seems clear and straightforward to the point 😅

andreondra, 2024-07-05:
Well, that's a good idea. You can try to generate some logos, if you get an interesting one, you can send it here and we'll decide what should we use 😁

andreondra, 2024-07-08:
Wow, these look very good. I like the first, fourth and fifth the most. Maybe we can pick one of those and use them as a organization logo as well?

mathieupotier, 2024-07-08:
I like the forth and fifth the most too...
I'll republish them later with alpha and proper cropping...

andreondra, 2024-07-08:
Thanks! :D

mathieupotier, 2024-07-16:
[image]

mathieupotier, 2024-07-16:
[image]

andreondra, 2024-07-16:
The second one (with the house) is probably the best. What do you think?

andreondra, 2024-07-16:
I cleaned up the artifacts and added a little bit of smoothing.

[image]

mathieupotier, 2024-07-16:
seems the most similar to the one we already have in orga ... suits well !! ^_^