sermatec-opensource/homeassistant-sermatec-inverter #13: Fix stuck values

andreondra, 2023-01-02:
Sometimes the integration would just stop updating the data and show only the last value (not Unavailable, which means the integration got stuck somehow).

andreondra, 2023-01-17:
This happens also in the latest version (2023.1.4).

andreondra, 2023-01-20:
It happened again. We should add logging to the data update coordinator and fix any bugs if needed. Weird thing is that it started to happen after a 2022.10 update, before that update it worked reliably.

andreondra, 2023-01-20:
Note: it could probably happen when the comm script returns empty values (connection error) — because no exception is raised, nothing shows up in the logs.

mathieupotier, 2023-02-12:
Hi @andreondra , got this error in logs, maybe related :
```
2023-02-12 15:27:30.101 ERROR (MainThread) [frontend.js.latest.202302020] :0:0 Script error.
2023-02-12 15:27:54.910 ERROR (MainThread) [custom_components.sermatec_inverter.sensor] Error fetching Sermatec data: Can't retrieve working parameters.
2023-02-12 15:27:54.914 ERROR (MainThread) [homeassistant.components.sensor] sermatec_inverter: Error on device update!
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 507, in _async_add_entity
    await entity.async_device_update(warning=False)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 739, in async_device_update
    await task
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 428, in async_update
    await self.coordinator.async_request_refresh()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 189, in async_request_refresh
    await self._debounced_refresh.async_call()
  File "/usr/src/homeassistant/homeassistant/helpers/debounce.py", line 82, in async_call
    await task
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 215, in async_refresh
    await self._async_refresh(log_failures=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 330, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 135, in async_update_listeners
    update_callback()
  File "/config/custom_components/sermatec_inverter/sensor.py", line 525, in _handle_coordinator_update
    self._attr_native_value = self.coordinator.data[self.dict_key]
TypeError: 'NoneType' object is not subscriptable
```

When getting an error from the inverter: trying to update an entity with bad type, seems that make the plugin to crash ...

gertz22, 2023-10-14:
Hello, same issue and log as mathieupotier, system aslo get too much log and can couse run out of fre space at home assistant which couse serious crash of server.

andreondra, 2024-04-11:
As with the other issue, this should be resolved in the latest version.

closed 2024-04-11 by andreondra