sermatec-opensource/homeassistant-sermatec-inverter #16: inversion on discharging and charging battery power

mathieupotier, 2023-01-09:
Hi there,

Battery is showing discharging power value instead of charging power value :
[image]

Formula for power in Direct Current context : P(Watts) = I(Ampers) * U(Volts)

Negative Watt value mean battery is charging
Positive Watt value mean battery is discharging.

Regards

andreondra, 2023-01-17:
That's weird, for me it shows the correct values with the current formula - negative current means the battery is discharging. It probably depends on the type of the battery, which the official app can check and then apply the correct formula. I have the Pylontech hi-voltage ones, what type do you use?

mathieupotier, 2023-01-18:
@andreondra
I have this material :
[image]

Maybe some setup on inverter, or firmware version related.

closed 2023-01-20 by andreondra