sermatec-opensource/homeassistant-sermatec-inverter #29: After installing the Home Assistant Core Update 2023.5.0, the integration stops working

MildaC, 2023-05-04:
After installing the Home Assistant Core Update 2023.5.0, the integration stops working. It does not load entities.

andreondra, 2023-05-04:
Thanks for the info, I'll have a look at it.

mathieupotier, 2023-05-04:
`ˋˋ2023-05-04 15:50:51.177 WARNING (SyncWorker_2) [homeassistant.loader] We found a custom integration sermatec_inverter which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2023-05-04 15:51:34.078 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry Sermatec Inverter for sermatec_inverter
File "/config/custom_components/sermatec_inverter/__init__.py", line 35, in async_setup_entry`ˋˋ

here are logs…

MildaC, 2023-05-04:
more info probably here https://github.com/home-assistant/core/pull/91929

andreondra, 2023-05-04:
Thanks guys, hopefully this weekend I'll have some time to fix this issue.

andreondra, 2023-05-07:
Platform loading should be changed to fix this issue: https://developers.home-assistant.io/docs/config_entries_index#for-platforms

PR #30 (2023-05-07): Removed deprecated async_setup_platforms.
Fixes #29
closed 2023-05-07 by andreondra