sermatec-opensource/homeassistant-sermatec-inverter #33: The problem of unavailability of sensors and meters

MildaC, 2023-05-10:
After updating Home Assistant Core to version 2023.5.2, there is an issue with the meters and the Riemann Integral Sum Sensor integration that retrieve states from the Sermatec integration. It seems that with zero input state, the meters and sensors are always unavailable at some house. I don't know if this is a bug in Home Assistant, or if the latest modification to the Sermatec integration may somehow be causing this. If I downgrade Home Assistant Core to version 2023.4.6, everything is fine. Can you please take a look at this? Thank you. I am attaching the log.

Logger: homeassistant.helpers.template_entity
Source: helpers/template_entity.py:344
First occurred: 08:43:42 (6 occurrences)
Last logged: 08:49:34

TemplateError('ValueError: Template error: round got invalid input 'unavailable' when rendering template '{{ (states('sensor.solar_inverter_battery_charging_power') | round(0)) - (states('sensor.solar_inverter_battery_discharging_power') | round(0)) }}' but no default was specified') while processing template 'Template<template=({{ (states('sensor.solar_inverter_battery_charging_power') | round(0)) - (states('sensor.solar_inverter_battery_discharging_power') | round(0)) }}) renders=34>' for attribute '_attr_native_value' in entity 'sensor.battery_power'
TemplateError('ValueError: Template error: round got invalid input 'unavailable' when rendering template '{{ (states('sensor.solar_inverter_grid_import') | round(0)) + (states('sensor.solar_inverter_battery_discharging_power') | round(0)) + (states('sensor.solar_inverter_pv_total_power') | round(0)) - (states('sensor.solar_inverter_battery_charging_power') | round(0)) - (states('sensor.solar_inverter_grid_export') | round(0)) }}' but no default was specified') while processing template 'Template<template=({{ (states('sensor.solar_inverter_grid_import') | round(0)) + (states('sensor.solar_inverter_battery_discharging_power') | round(0)) + (states('sensor.solar_inverter_pv_total_power') | round(0)) - (states('sensor.solar_inverter_battery_charging_power') | round(0)) - (states('sensor.solar_inverter_grid_export') | round(0)) }}) renders=270>' for attribute '_attr_native_value' in entity 'sensor.okamzita_spotreba_domu'
TemplateError('ValueError: Template error: round got invalid input 'unavailable' when rendering template '{{ (states('sensor.solar_inverter_battery_charging_power') | round(0)) - (states('sensor.solar_inverter_battery_discharging_power') | round(0)) }}' but no default was specified') while processing template 'Template<template=({{ (states('sensor.solar_inverter_battery_charging_power') | round(0)) - (states('sensor.solar_inverter_battery_discharging_power') | round(0)) }}) renders=58>' for attribute '_attr_native_value' in entity 'sensor.battery_power'
TemplateError('ValueError: Template error: round got invalid input 'unavailable' when rendering template '{{ (states('sensor.solar_inverter_grid_import') | round(0)) + (states('sensor.solar_inverter_battery_discharging_power') | round(0)) + (states('sensor.solar_inverter_pv_total_power') | round(0)) - (states('sensor.solar_inverter_battery_charging_power') | round(0)) - (states('sensor.solar_inverter_grid_export') | round(0)) }}' but no default was specified') while processing template 'Template<template=({{ (states('sensor.solar_inverter_grid_import') | round(0)) + (states('sensor.solar_inverter_battery_discharging_power') | round(0)) + (states('sensor.solar_inverter_pv_total_power') | round(0)) - (states('sensor.solar_inverter_battery_charging_power') | round(0)) - (states('sensor.solar_inverter_grid_export') | round(0)) }}) renders=320>' for attribute '_attr_native_value' in entity 'sensor.okamzita_spotreba_domu'
TemplateError('ValueError: Template error: round got invalid input 'unavailable' when rendering template '{{ (states('sensor.solar_inverter_battery_charging_power') | round(0)) - (states('sensor.solar_inverter_battery_discharging_power') | round(0)) }}' but no default was specified') while processing template 'Template<template=({{ (states('sensor.solar_inverter_battery_charging_power') | round(0)) - (states('sensor.solar_inverter_battery_discharging_power') | round(0)) }}) renders=72>' for attribute '_attr_native_value' in entity 'sensor.battery_power'

Akstef, 2023-05-21:
I am new in home assistant and I have no programmation skills. I use the sermatec integration in home assistant 2023.5.3 and after several minutes all sensors becaume unavailable. I need to reload the integration to have the data back again. Is it a bug or should I change something in my configuration?

andreondra, 2023-05-21:
It could be a bug but also it can be caused by the inverter itself which sometimes stops responding to the requests.

Akstef, 2023-05-21:
I don't think it is the inverter which stops responding. If I am working one hour to configure my system and I am connected to it, the sensors never stop. If only my smartphone is used to check from time to time how the system react, after 20 to 30 minutes of inactivity, the sensors stop working. As soon as I reload the integration, all sensors are back. Can you integrate an auto reload fonction when the sensors become unavailable?

MBoronNeotreks, 2023-05-21:
Maybe adding more try/except or specifying call timeouts to the `sermatec-inverter` project could help.
I see these are used only for the `connect` function.
But I checked that project just briefly and I'm not a Python programmer.

mboron, 2023-05-21:
Maybe adding more try/except or specifying call timeouts to the `sermatec-inverter` project could help.
I see these are used only for the `connect` function.
But I checked that project just briefly and I'm not a Python programmer.
Btw. I'm facing the very same issue.

mathieupotier, 2023-05-25:
having few logs in my instance that could explain the error :
```
2023-05-25 19:43:43.100 ERROR (MainThread) [custom_components.sermatec_inverter.sensor] Error fetching Sermatec data: Can't connect to the inverter.
2023-05-25 19:43:43.103 ERROR (MainThread) [homeassistant.components.sensor] sermatec_inverter: Error on device update!
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 521, in _async_add_entity
    await entity.async_device_update(warning=False)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 779, in async_device_update
    await self.async_update()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 444, in async_update
    await self.coordinator.async_request_refresh()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 211, in async_request_refresh
    await self._debounced_refresh.async_call()
  File "/usr/src/homeassistant/homeassistant/helpers/debounce.py", line 95, in async_call
    await task
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 237, in async_refresh
    await self._async_refresh(log_failures=True)
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 349, in _async_refresh
    self.async_update_listeners()
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 146, in async_update_listeners
    update_callback()
  File "/config/custom_components/sermatec_inverter/sensor.py", line 525, in _handle_coordinator_update
    self._attr_native_value = self.coordinator.data[self.dict_key]
TypeError: 'NoneType' object is not subscriptable
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 146, in async_update_listeners
    update_callback()
  File "/config/custom_components/sermatec_inverter/sensor.py", line 525, in _handle_coordinator_update
    self._attr_native_value = self.coordinator.data[self.dict_key]
TypeError: 'NoneType' object is not subscriptable
```

mathieupotier, 2023-05-25:
@Akstef the inverter sometimes stop responding, having the same issue on another project.

I assume that the connected socket die, and need to reconnect to the inverter to have queries and response back again.
But sometimes for some reason, the inverter drops the TCP socket... 🤷‍♂️ 
Need to be taken into account when developping an inverter client...
But the protocol itself is quite magic ... already amazing to have a home assistant plugin for that one ... cheers 🍾 @andreondra

mathieupotier, 2023-05-25:
Created an automation in my HA instance to try to reload the integration automatically (better thazn nothing) ... but sometimes, it just fail endlessly ... 🤷‍♂️ 

```yaml
alias: Reload auto Sermatec
description: ""
trigger:
  - platform: state
    entity_id:
      - sensor.solar_inverter_inverter_working_mode
      - sensor.solar_inverter_battery_state
      - sensor.sermatec_solar_inverter
    to: unavailable
    for:
      hours: 0
      minutes: 0
      seconds: 30
  - platform: state
    entity_id:
      - sensor.solar_inverter_inverter_working_mode
      - sensor.solar_inverter_battery_state
      - sensor.sermatec_solar_inverter
    to: unknown
    for:
      hours: 0
      minutes: 0
      seconds: 30
condition: []
action:
  - repeat:
      while:
        - condition: or
          conditions:
            - condition: state
              entity_id: sensor.sermatec_solar_inverter
              state: unknown
              for:
                hours: 0
                minutes: 0
                seconds: 0
            - condition: state
              entity_id: sensor.solar_inverter_battery_state
              state: unknown
              for:
                hours: 0
                minutes: 0
                seconds: 0
            - condition: state
              entity_id: sensor.solar_inverter_inverter_working_mode
              state: unknown
              for:
                hours: 0
                minutes: 0
                seconds: 0
            - condition: state
              entity_id: sensor.sermatec_solar_inverter
              state: unavailable
              for:
                hours: 0
                minutes: 0
                seconds: 0
            - condition: state
              entity_id: sensor.solar_inverter_battery_state
              state: unavailable
              for:
                hours: 0
                minutes: 0
                seconds: 0
            - condition: state
              entity_id: sensor.solar_inverter_inverter_working_mode
              state: unavailable
              for:
                hours: 0
                minutes: 0
                seconds: 0
      sequence:
        - service: homeassistant.reload_config_entry
          data: {}
          target:
            entity_id: sensor.sermatec_solar_inverter
        - delay:
            hours: 0
            minutes: 0
            seconds: 15
            milliseconds: 0
mode: single
```

andreondra, 2023-05-25:
Hello @mathieupotier, the reload automation looks good. 👍 Maybe you can add it to the readme as a temporary fix?

I am aware of the poor state of the communication script. Now I have a lot of work with finishing my degree but after the examination I'll have some time to rewrite the script and add more features to the integration.

(Or if someone else fixes the issue earlier than me, I'll be happy to merge a PR of course.)

Akstef, 2023-06-05:
I have maybe a stupid remark. 
If the inverter stops sometimes the communication on the wifi network, is it not possible to install a RS485 to ethernet converter and get all the sensors data over ethernet?
I have checked the python files to understand how it works but it is definitely over my knowledge.

andreondra, 2023-06-05:
Hello @Akstef , it is definitely possible. The Modbus communication is usually used to get data for Wattrouters and similar devices. You can't read everything via Modbus but all the basic stuff yes.

Please have a look at [https://github.com/sermatec-opensource/docs](this repo) where I documented the RS485 Modbus protocol.

Akstef, 2023-07-16:
Hello @andreondra. I have checked the documentation but I am definitely not able to program something by myself but I find a way to use your integration "without" crash.

@mathieupotier has created an automation but sometimes the integration stop to work. I have created another one. With my automation I have the 24 hours displayed in the energy dashboard. The integration still continue to crash from time to time but it always restart. The datas collected for one day are almost correct (I loose only the reboot time).

Here is my automation if it can help someone:

alias: Inverter restart
description: ""
trigger:
  - platform: time_pattern
    seconds: /30
condition:
  - condition: or
    conditions:
      - condition: state
        entity_id: sensor.solar_inverter_battery_state
        state: unavailable
      - condition: state
        entity_id: sensor.solar_inverter_battery_state
        state: unknown
action:
  - repeat:
      count: "3"
      sequence:
        - service: homeassistant.reload_config_entry
          data: {}
          target:
            device_id:  of the sermatec integration
  - delay:
      hours: 0
      minutes: 0
      seconds: 5
      milliseconds: 0
  - if:
      - condition: or
        conditions:
          - condition: state
            entity_id: sensor.solar_inverter_battery_state
            state: unavailable
            for:
              hours: 0
              minutes: 0
              seconds: 0
          - condition: state
            entity_id: sensor.solar_inverter_battery_state
            state: unknown
            for:
              hours: 0
              minutes: 0
              seconds: 0
    then:
      - service: homeassistant.restart
        data: {}
mode: single

Another question. Why is the integration looking for the data on the STA IP address? Is it no possible to use the AP IP address (10.10.10.100) so we don't need to send the data to the sermatec cloud and keep everything in local?

Quipfr, 2023-09-16:
Hi,
Since one week, I had many issues with HA : all sensor of sermatec are always "unvailable". I tried to restore some old realase wtithout any sucess, I download the last files and make the automation for reaload the service of sermatec. The fix is good only when I'm on the web interface of HA but when I close it, the automation doesn't launch. So I lost all my optimisation when my batteries are full.
I'm on those version : 
HA/supervisor 2023.09.2
OS 10.5
User Interface 20230911.0

Akstef, 2023-09-16:
@Quipfr 
- Did you restart fully your HA instance? Not only the yaml configuration
- Do you have the inverter data on sermatec cloud? It happends sometime to me that the inverter lost the network parameters. It change to AP mode instead of STA mode
- I don't understand the issue of the automation. The automation starts every 30s to check if the inverter is available or not. It is fully independant of the location of your HA instance

Quipfr, 2023-09-16:
@Akstef, I restart fully my HA 5 times each day :)
I don't have the inverter data on sermatec cloud (I"m not very confident with that)
I agree with you about the automation, but it does'nt work like that. I'm currently check it in the history and the automation launched only when I'm connected at my HA instance :(
Many thanks for your help

Akstef, 2023-09-16:
I was not able to get any data in HA if the STA mode was not activated. If I used the AP IP adress 10.10.10.100 it is not working at all. I needed to activate the STA mode and therefore send the data over the cloud.

Yes I changed the sermatec inverter to fronius and victron after the sermatec burned mid July. I had a shortcut on a light and the inverter burned when the light safety has tripped. 
Fronius and victron work fine. More expensive and harder to configure but once done it is good

andreondra, 2023-09-16:
Hello, thank you for your comments and suggestions.

I have almost finished a new version of the script. Hopefully it'll fix these issues so no more workarounds are needed. Also it'll introduce some new features.

I checked the state of the official app and it's laughable, so I hope the new version of this Hass component will make some owners a little bit happier.

Quipfr, 2023-12-03:
Hello Andreondra,
I'm posting on the forum again because my problem remains unresolved.
Thank you in advance for your feedback and thanks again for all the work you've done. :)

andreondra, 2024-04-11:
Hello all, this should be resolved in the latest version. If similar problem appears again, open a new issue.

closed 2024-04-11 by andreondra

mboron, 2024-04-12:
Ondra, thank you for all your hard work on this project!

Akstef, 2024-04-12:
Thank you for the new version but I will not be able to test it because I changed my system but it is good to see you made an update as promise. Thank you again and take care.