sermatec-opensource/homeassistant-sermatec-inverter #36: Have last inverter contact date and time

mathieupotier, 2023-10-31:
Hi,

Sometimes the inverter doesn't answer anymore, I propose to add a date time updated when we successfully have the inverter responding. Allowing us to detect when we're having trouble and then reload the integration and have data back again.

andreondra, 2024-04-05:
Hi, thanks for your idea. I may add this to a new version of the integration, but hopefully this won't be needed -- it looks like the new version (now available in the `dev` branch) is way more reliable. I added a bunch of error checking, retries etc. to the communication script.

andreondra, 2024-04-25:
For now I close this. If some problems arise in the future, we can reconsider this feature.

closed 2024-04-25 by andreondra