sermatec-opensource/homeassistant-sermatec-inverter #45: Bug : UnicodeDecodeError: 'ascii' codec can't decode byte 0xfe in position 0: ordinal not in range(128)

mathieupotier, 2024-04-12:
Hi,

Just passed to the "latest" version, got thisd error which has appeared in logs

```
Enregistreur: custom_components.sermatec_inverter.coordinator
Source: helpers/update_coordinator.py:315
intégration: Sermatec Inverter (documentation)
S'est produit pour la première fois: 12:32:29 (3 occurrences)
Dernier enregistrement: 12:33:29

Unexpected error fetching Sermatec data: 'ascii' codec can't decode byte 0xfe in position 0: ordinal not in range(128)
Unexpected error fetching Sermatec data: 'ascii' codec can't decode byte 0x8f in position 0: ordinal not in range(128)
Unexpected error fetching Sermatec data: 'ascii' codec can't decode byte 0xa4 in position 0: ordinal not in range(128)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 315, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/sermatec_inverter/coordinator.py", line 59, in _async_update_data
    response = await self.smc_api.getCustom(cmd)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/sermatec_inverter/sermatec_inverter/__init__.py", line 195, in getCustom
    parsedResponse.update(self.parser.parseReply(responseCode, self.pcuVersion, response))
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/sermatec_inverter/sermatec_inverter/protocol_parser.py", line 374, in parseReply
    newField["value"] = trimmedString.decode('ascii')
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
UnicodeDecodeError: 'ascii' codec can't decode byte 0xfe in position 0: ordinal not in range(128)
```

andreondra, 2024-04-12:
Thanks for the report, I'll have a look at it.

andreondra, 2024-04-13:
This is caused by a bug in protocol parser, inverter returns reply with wrong length and command code but the parser proceeds with parsing nevertheless.

PR #49 (2024-04-17): Fixed data querying
The PR should fix #45 and fix #46
closed 2024-04-17 by andreondra