sermatec-opensource/homeassistant-sermatec-inverter #46: Bug : IndexError: string index out of range

mathieupotier, 2024-04-12:
Hi all,
Just passed to "latest" version and got this error appearing in logs.

```
Enregistreur: custom_components.sermatec_inverter.coordinator
Source: helpers/update_coordinator.py:315
intégration: Sermatec Inverter (documentation)
S'est produit pour la première fois: 12:31:58 (2 occurrences)
Dernier enregistrement: 12:33:59

Unexpected error fetching Sermatec data: string index out of range
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 315, in _async_refresh
    self.data = await self._async_update_data()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/sermatec_inverter/coordinator.py", line 59, in _async_update_data
    response = await self.smc_api.getCustom(cmd)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/sermatec_inverter/sermatec_inverter/__init__.py", line 195, in getCustom
    parsedResponse.update(self.parser.parseReply(responseCode, self.pcuVersion, response))
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/sermatec_inverter/sermatec_inverter/protocol_parser.py", line 377, in parseReply
    newField["value"] = bool(binString[fieldBitPosition])
                             ~~~~~~~~~^^^^^^^^^^^^^^^^^^
IndexError: string index out of range```

andreondra, 2024-04-12:
Hi, thanks for the report. Can you please turn on the debugging and paste the logs? So I can see what exactly the inverter returned. The log may be too large to paste here, so probably paste only the relevant part here or send the whole log via e-mail/Google Drive etc. Thank you

mathieupotier, 2024-04-12:
I'll activate debug mode for the WE end give you all relevant logs on next week ^_^

PR #49 (2024-04-17): Fixed data querying
The PR should fix #45 and fix #46
closed 2024-04-17 by andreondra