sermatec-opensource/homeassistant-sermatec-inverter #50: Error unit type reactive and apparent power

geof6298, 2024-04-17:
Hello,

il y a une erreur sur la puissance réactive qui est en VAR et non en W.
sur la puissance apparente qui est en VA et non en VAR.

une petite correction :-)

Bonne Journée

mathieupotier, 2024-04-17:
hi @geof6298

mathieupotier, 2024-04-17:
[image]

geof6298, 2024-04-17:
dans le fichier protocole-en.json
[image]

et sur mon tableau de bord

[image]

mathieupotier, 2024-04-17:
ah ok ... dsl j'avais pas compris ... ^_^

geof6298, 2024-04-17:
je me suis peut être mal exprimé aussi lol

mathieupotier, 2024-04-17:
@andreondra 

Seems that there is a mistake on unit used in protocol-en.json
Should it be fixed in the file ? or it is generated from the sermatec one ?

mathieupotier, 2024-04-17:
@andreondra Seems that it comes with error even in original repository ...

andreondra, 2024-04-17:
It can be fixed in the file, it's not generated. It's probably a mistake present also in the app :D

mathieupotier, 2024-04-17:
this repo will be the future truth source for sermatec ... ^_^

PR #52 (2024-07-05): Comm script sync
Fixes #51
Fixes #50
Fixes #48
closed 2024-07-05 by andreondra