sermatec-opensource/homeassistant-sermatec-inverter #60: Test parameter setting

andreondra, 2024-07-16:
Hi @mathieupotier,

finally, it is possible to configure basic parameters by this integration. When you have time, can you please test this feature on your inverter? If you encounter no errors, we can finally release a v2.1.0! Note that there will be slight delay (up to 1 min) between toggling the value and state update... This is expected and described in README.

mathieupotier, 2024-07-16:
@andreondra assign this issue to me ...
I'll try to deploy this on my installation in the next few days ...

mathieupotier, 2024-07-16:
deployed this branch on my integration, seems ok, I'll let it run for a couple of hours, and let you know if I had any error message in logs

mathieupotier, 2024-07-17:
seems ok, no significant errors raised after 24h

andreondra, 2024-07-18:
Great! We are ready for new release then.

closed 2024-07-18 by andreondra